jellewie/Arduino-Smart-light #25: AMS1117-3.3V is getting hot

jellewie, 2020-08-14:
The voltage regulator on the ESP32 is getting really hot (PCB V1.2, 12V input) even without the data line for the LEDs attached

jellewie, 2020-08-14:
It's a linear voltage regulator, to it will burn away 12-3.3V times the current. It does work, but the life expectancy is just lower.

Maybe look into LM2596S or other switching regulators?

jellewie, 2020-08-17:
For 12V there needs to be an LM2596 in the schematic, for 5V it is fine as is.
So add 'LM2596' as an option to the PCB

closed 2023-03-05 by jellewie

harbaum, 2023-06-21:
The AMS1117 family is produced by many manufacturers. A lot of these (especially the ones rated for a higher current) are limited to an input voltage of 12V. In my experience especially the very cheap ones are really limited to 12V. They work well up to slightly over 11V. But at 12V they die even at very low currents at their output.

This can e.g. observed with many cheap arduino nano clones.