jellewie/Arduino-Smart-light #53: Home Assistant integration

jellewie, 2023-03-05:
I want to integrate the lamp into Home Assistant, but don't know exactly how yet.

For now you can use this, where "192.168.1.255" is the IP of the lamp:
```
light:
  - platform: template
    lights: #https://www.home-assistant.io/integrations/light.template/
      smartclock:
        turn_on:
          service: rest_command.smartclock_on
        turn_off:
          service: rest_command.smartclock_off
        set_level:
          service: rest_command.smartclock_brightness
          data:
            brightness: "{{ brightness }}"
rest_command: #https://www.home-assistant.io/integrations/rest_command/
  smartclock_on:
    url: http://192.168.1.255/set?m=ON&r=255&g=200&b=100&s=2
    verify_ssl: false
  smartclock_off:
    url: http://192.168.1.255/set?m=clock&1=true
    verify_ssl: false
  smartclock_brightness:
    url: http://192.168.1.255/set?l={{brightness}} #https://community.home-assistant.io/t/switch-with-rest-command/154762/2
    verify_ssl: false
```
[image]

Currently, it's configured with "s=2" with means it only does the not-clock section. Remove the "&s=2" to make it change all LEDs. See 
https://github.com/jellewie/Arduino-Smart-light/blob/ff45a61ef0f196b925e764d9e2b6ba6c03132d79/Arduino/handler.ino#L23

Currently, it's set up to change the RGB with "r=255&g=200&b=100", but your own settings can be put in. no RGB support yet (HA withs with Hue/saturation and I haven't been bordered to figure that out yet.

wjtje, 2023-03-06:
Maybe the best way to integrate your own code into Home Assistant is using MQTT.

Your code can send an MQTT Discovery to Home Assistant so it will be auto discovered and configured ([docs](https://www.home-assistant.io/integrations/mqtt/#mqtt-discovery)), and it can send state changes to Home Assistant.

jellewie, 2024-02-28:
Got a lot of it working in MQTT, but Still some things to fix/do

- [ ] can byte mac[] be not-static?
- [ ] fix setConfigurationUrl
- [ ] when onStateCommand1(OFF), let it change back the mode. and restore AutoBrightness?
- [ ] Update the (other) light to HA at the right moments
- [ ] How to send RGB data to HA? the color is sometimes a mismatch
- [ ] How to send Brightness data to HA? the Brightness is sometimes a mismatch